STMicroelectronics/STM32CubeF1 #3: update to cmsis os2

mgiaco, 2019-07-09:
Hi,

Are there any plans to update to a new freertos and cmsis os2?

many thanks
cheers
mathias

CCASTM, 2019-07-10:
Hello Mathias,
The next release STM32CubeF1 FW Package (V1.8.0), expected in the second half of July 2019 will include the V10.0.1 FreeRTOS MW, including CMSIS-OS-V2
Regards
Christophe

mgiaco, 2019-07-16:
okay that sounds really great i hope it will be published here.
Best Regards,
mathias

mgiaco, 2019-07-22:
Many thanks for the update, the problem is that it is not possible to switch to CMSIS-OS-V2 in cube mx but I think that is a general issue.  Is there a roadmap for CubeMX ?  I found a issue for STM32F1 and ADC Code gerneration. Where is the right place to report that?

jecaille, 2019-07-25:
Hi Mathias,
Thanks for reporting.
Please go to the st community group stm32mcu :
https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus
Best regards,
Jérôme

CCASTM, 2019-08-12:
Hello Mathias,
I confirm that for the moment, you can use the FW PAckage as is (CMSIS and FreRTOS version have been updated) but the integration in CubeMX. is not yet available.
In the context of 13 different STM32 series, we are still working on the priority list for CubeMX, and the plan is not yet defined.
Regards
Christophe

ALABSTM, 2019-09-16:
Hi Mathias,

I hope you are doing well. As this issue is now related to CubeMX but not to the firmware published into this repository, would you agree to close it? Thank you in advance for your answer.

As you may know issues are exclusively reserved to aspects directly related to the firmware published into GitHub repository. Any other question could be submitted to the [ST Community](https://community.st.com/s/) and particularly under ["STM32CubeMX"](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) topic in case of a question as yours.

With regards,

closed 2019-09-17 by mgiaco

mgiaco, 2019-09-25:
@ALABSTM 
Is there any way to report an issue for cubemx in any ticket system. Because I have the problem that for me i looks like no one is looking on a issue when reporting on ST Community. As an example i have reported that there is any problem when using ADC3 on F1 .... but nothing is happend. So no comment or hint or what else...
https://community.st.com/s/question/0D50X0000AYtJRaSQN/latest-cubemx-and-latest-cubef1-generates-wrong-code-f%C3%BCr-adc3-with-ll-and-dma-can-someone-also-has-the-same-problem

many thanks

ALABSTM, 2019-09-26:
Hi Mathias,

As far as I know, CubeMX-related issues are indeed reported on the ST Community. I just checked your post and found that someone has finally answered. We do apologize for the delays and we do our best to improve our response time.

Now, regarding posting on the ST Community, topics have been created in order to better classify the posts received and to answer them more rapidly.

As for CubeMX-related issues, the dedicated topic is [STM32CubeMX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx). As you may notice, around 740 posts are listed under this topic till now. On the other hand, topic [STM32 MCUs](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus) counts more than 7k posts. This makes the task longer to identify priority posts under this topic and answer them.

The most important now is that your request has been logged into our internal bug tracking system. Hopefully you receive an answer soon and have this issue fixed within a future CubeMX release.

With regards,

mgiaco, 2019-10-01:
Okay many thanks