STMicroelectronics/STM32CubeF1 #4: Multiple warning about unused parameters.

TByte007, 2019-09-26:
Is it possible to put "attribute ((unused))" before those params because it's annoying and fills the screen with useless info and I really want to be able to use the -Wall parameter.

> In file included from C:\Users\tbyte\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\cores\arduino\stm32\HAL\stm32yyxx_hal_tim_ex.c:5:
> C:\Users\tbyte\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim_ex.c: In function 'HAL_TIMEx_RemapConfig':
> C:\Users\tbyte\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim_ex.c:1734:60: warning: unused parameter 'htim' [-Wunused-parameter]
>  HAL_StatusTypeDef HAL_TIMEx_RemapConfig(TIM_HandleTypeDef *htim, uint32_t Remap)
>                                          ~~~~~~~~~~~~~~~~~~~^~~~
> C:\Users\tbyte\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim_ex.c:1734:75: warning: unused parameter 'Remap' [-Wunused-parameter]
>  HAL_StatusTypeDef HAL_TIMEx_RemapConfig(TIM_HandleTypeDef *htim, uint32_t Remap)
>                                                                   ~~~~~~~~~^~~~~
> In file included from C:\Users\tbyte\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\cores\arduino\stm32\usb\usb_device_ctlreq.c:3:
> C:\Users\tbyte\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_ctlreq.c: In function 'USBD_CtlError':
> C:\Users\tbyte\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_ctlreq.c:853:42: warning: unused parameter 'req' [-Wunused-parameter]
>                     USBD_SetupReqTypedef *req)
>                     ~~~~~~~~~~~~~~~~~~~~~~^~~
>

ALABSTM, 2019-09-26:
Hi TByte007,

Our development teams confirmed the issue you pointed out. The fix will be available in a future release. Thank you for your contribution.

With regards,

ALABSTM, 2019-09-26:
ST Internal Reference: 72918

ALABSTM, 2019-09-26:
ST Internal Reference: 72921

ALABSTM, 2019-12-12:
Hi Tbyte007,

A fix to prevent the warning you are getting has been implemented on both TIM and USB drivers. It will be made available in the next STM32CubeF1 release. Unfortunately we cannot share a date for the moment.

Here is how it has been handled in the case of the TIM:
```diff
HAL_StatusTypeDef HAL_TIMEx_RemapConfig(TIM_HandleTypeDef *htim, uint32_t Remap)
{
+  /* Prevent unused argument(s) compilation warning */
+  UNUSED(htim);
+  UNUSED(Remap);

  return HAL_OK;
}
```
I hope this answer helps. If you have no other question about this subject, I think this issue can be closed.

With regards,

thomask77, 2020-01-16:
Similar warnings occur in https://github.com/STMicroelectronics/STM32CubeH7

Will it be fixed there also? Do we have to file another bug report?

best regards,
Thomas

ALABSTM, 2020-02-03:
Hi Thomask77,

Regarding the STM32CubeH7 firmware, the warning does not exist for the TIM function. The issue regarding the USB function, however, exists indeed as you mentioned.

This latter has been fixed in the USB Device Library v2.6.0. This new library revision shall be integrated to the next firmware revision (v1.7.0, normally) or to a later one. Hence, no need to file a new bug report, I think.

With regards,

ALABSTM, 2020-12-02:
Hi @TByte007,

I hope you are fine. You request has been addressed at the level of the TIM driver.

For the USB device library, the change has been implemented in the frame of version v2.6.0. Unfortunately, for the STM32CubeF1 there is no plan to upgrade from **v2.5.3**  to **v2.6.0**.

Please allow me to close this issue now and thank you for your comprehension.

With regards,

closed 2020-12-02 by ALABSTM

ALABSTM, 2021-06-21:
Hi @TByte007,

I came back to this issue and realized I did not provide you with the link to the [stm32_mw_usb_device](https://github.com/STMicroelectronics/stm32_mw_usb_device) repository where the USB Device library is published as a separate component.

For information, all **v2.\*.\*** versions are **backward compatible**. You can therefore even consider the use of the latest version **v2.8.0** published a couple of months ago. I hope this helps.

With regards,

TByte007, 2021-08-02:
> Hi @TByte007,
>
> I came back to this issue and realized I did not provide you with the link to the [stm32_mw_usb_device](https://github.com/STMicroelectronics/stm32_mw_usb_device) repository where the [...]

I'm sorry  I was busy with a different project which took all my time. Thank you for your time and effort ! I'll test it as soon as I have some spare time.