STMicroelectronics/STM32CubeF1 #5: Broken reset handler for SMT32F103xE?

jimvonmoon, 2019-10-27:
I think reset handler in `Drivers/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/startup_stm32f103xe.s` might be missing code responsible for initializing stack pointer.
```
Reset_Handler:
  ldr sp, =_estack    /* <=== Missing? */

/* Copy the data segment initializers from flash to SRAM */
  movs r1, #0
  b LoopCopyDataInit
```
I created a project for STM32F103ZET6 in STM32CubeMX (apparently up-to-date) and right away I noticed `malloc()` failing for larger buffer sizes (like 1kB). I noticed that `_sbrk()` was failing because of stack collision check. Turned out that SP register was set to near beginning of SRAM block (something like `0x2000014a`, basically right after BSS section). After adding this one line to reset handler everything started working nicely.

Is this how it's supposed to work? I am totally new to all this, so feel free to ignore this report. I just thought I'd let you know of potential problem. Thanks!

ALABSTM, 2019-10-29:
Hi Jim Von Moon,

Thank you for having sent this issue and, of course, we will take it into consideration. It will be forwarded to our development teams to check whether it is relevant or not.

We will be back to you as soon as we get an answer. So stay tuned and thank you in advance for your patience.

With regards,

mdmikh, 2019-10-30:
Actually first interrupt vector is used to set initial sp value so no explicit initialization is necessary.

jimvonmoon, 2019-10-31:
Yes, I think that's how it's supposed to work, but just a minute ago I created a new project in STM32CubeMX for STM32F103ZET6. The only thing I changed was enabling Serial Wire to be able to debug. Right after entering `main()` SP register was set to `0x200001f4`. After manually setting SP register to `_estack` in reset handler it's value is `0x2000fff8`.

ALABSTM, 2020-01-15:
Hi JimVonMoon,

I hope you are doing well. Please excuse the delay it took, but we finally got a feedback from our development teams.

Below are the details, hoping it answers your question:
* Stack pointer initialization
  The vector table contains the reset values of stack pointer and reset handler.
  After reset, the vector table is mapped at address 0, and the SP register is initialized with the same value as in the vector table.
  So technically there is no need to initialize the SP unless:
  * The Reset_Handler is explicitly called in the code.
  * The debugger is performing an emulated reset by setting the PC to the reset handler and forgetting about the SP default value.

  References:
  * http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0552a/BABIFJFG.html
  * http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0552a/Ciheijba.html

* Stack/Heap collision
  STM32F103ZE has 64K RAM. this means 0x2000014a as SP value is far from the end of RAM (0x20010000).
  Having a collision means either:
  * The user functions are using too much local variables.
  * The _estack value is wrong.

  An easy test could be to check the generated linked file (.ld) by CubeMX.

Please do not hesitate if you need more details or support. I let @tarek-bochkati answer you upcoming questions in the frame of this thread.

With regards,

jimvonmoon, 2020-02-08:
Hi again. Sorry for the delayed response, it was only recently I got some time to get back to my project. Thank you very much for the detailed answer. It seems in my case it was a problem with the debugger. Everything seems to run fine without it, no need to set SP manually. Thank you for help!

closed 2020-02-08 by jimvonmoon

tgree, 2020-07-04:
Actually, @ALABSTM that is not entirely correct.  The SP is *supposed* to be initialized based on the first interrupt vector entry.  However, in practice ST doesn't support that for all choices of address.  The ST boot loader validates the address first, and for some chips that have SRAMs aliased at both low (0x10000000) and high (0x20000000) addresses, the boot loader doesn't always support using the low addresses in the vector table.

See the following thread in community.st for more details of how I dealt with this and with how the __stack variable in the GCC startup script was very helpful in resolving the issue:

https://community.st.com/s/question/0D50X0000CEr55SSQR/bfb2-with-exception-vector-stack-in-ccm-sram

ALABSTM, 2020-07-22:
Hi @tgree,

Thank you for this comment. I read the analysis you posted on the community. It was very instructive. Thank you for the sharing.

With regards,

ALABSTM, 2020-07-29:
Hi @tgree,

Just to let you know, I forwarded you message and the post you shared on the community to our development teams. They said what you reported regarding the STM32G474CET microcontroller is a limitation of the  bootloader itself they are already aware of. It shall be reported in [this](https://www.st.com/resource/en/application_note/cd00167594-stm32-microcontroller-system-memory-boot-mode-stmicroelectronics.pdf) application note.

With regards,