STMicroelectronics/STM32CubeF1 #8: tranmit -> transmit typo

martonmiklos, 2019-11-21:
Here:
https://github.com/STMicroelectronics/STM32CubeF1/blob/441b2cbdc25aa50437a59c4bffe22b88e78942c9/Drivers/STM32F1xx_HAL_Driver/Inc/stm32f1xx_hal_can.h#L298-L302

There is a tranmit -> transmit typo in the comment. See PR #7 for details.
```

-#define ...  /*!< ... due to tranmit error    */
+#define ...  /*!< ... due to transmit error    */
```

ALABSTM, 2019-11-21:
Hi Miklòs,

Thank you for your collaboration. I took the initiative to add the "diff" tag so that differences are correctly displayed in your comment.

Your remark will be transmitted to our development teams to have this typo fixed in the frame of a future release. Thank you once more for this contribution.

With regards,

martonmiklos, 2019-11-21:
I was not aware of this code-diff markdown feature, so I learned something new again! Thanks for the hint.

ALABSTM, 2019-11-25:
ST Internal Reference: 76404

closed 2020-10-12 by ALABSTM