STMicroelectronics/STM32CubeF1 #9: HAL_PCD_EP_GetRxCount returns incorrect value for multi-packet transfers

a-v-s, 2020-01-18:
**Describe the set-up**
 I'm using a Blue Pill STM32F103C8T6
I'm running arm-none-eabi-gcc (Arch Repository) 9.2.0
I am using the code from this repository

**Describe the bug**
HAL_PCD_EP_GetRxCount returns incorrect value for multi-packet transfers. It returns the value for the last packet in the transfer.

stm32f1xx_hal_pcd.c line 2142 increases the value
`ep->xfer_count += count;`
however, when there is another paket to receive on line 2156 it calls
`(void)HAL_PCD_EP_Receive(hpcd, ep->num, ep->xfer_buff, ep->xfer_len);`
then on line 1632, the HAL_PCD_EP_Receive function resets the ep->xfer_count
`HAL_PCD_EP_Receive`

**How To Reproduce**
Create a packet that uses the HAL USB drivers, then initiate an OUT transfer with a length greater then the endpoint size. 

**Additional context**
A possible fix is to uncomment the resetting of the xfer_count in the HAL_PCD_EP_Receive

```
        if ((ep->xfer_len == 0U) || (count < ep->maxpacket))
        {
          /* RX COMPLETE */
#if (USE_HAL_PCD_REGISTER_CALLBACKS == 1U)
          hpcd->DataOutStageCallback(hpcd, ep->num);
#else
          HAL_PCD_DataOutStageCallback(hpcd, ep->num);
#endif /* USE_HAL_PCD_REGISTER_CALLBACKS */
          ep->xfer_count = 0; // Reset the xfer_count here
        }

```

 and adding it after the DataOutStageCallback call. 

```
 /*setup and start the Xfer */
  ep->xfer_buff = pBuf;
  ep->xfer_len = len;
  //ep->xfer_count = 0U; // in stead of here
  ep->is_in = 0U;
  ep->num = ep_addr & EP_ADDR_MSK;
```

ALABSTM, 2020-02-03:
Hi André,

Thank you for having reported this point and thank you for all these details you provided. Please allow me to introduce @bouattay from our USB development teams who will answer your post.

With regards,

bouattay, 2020-02-05:
Hi André,

Yes I can confirm the reported issue, could you please check this proposal to avoid resetting xfer_count  

        else
        {
          //(void)HAL_PCD_EP_Receive(hpcd, ep->num, ep->xfer_buff, ep->xfer_len);
          (void)USB_EPStartXfer(hpcd->Instance, ep);
        }

Thanks,
Aymen

a-v-s, 2020-02-05:
The proposed change appears to work fine. 

Thanks,
André

ALABSTM, 2020-03-04:
ST Internal Reference: 82061

a-v-s, 2020-03-04:
The problem occurs on the F0 libraries as well.

closed 2020-10-12 by ALABSTM