STMicroelectronics/STM32CubeF1 #12: extern uwTick : Missing from F1 variant stm32f1xx_hal.h

geosmall, 2020-05-14:
First raised here https://github.com/stm32duino/Arduino_Core_STM32/issues/1064#issuecomment-627340722

**Describe the bug** : extern __IO uint32_t uwTick;
This declaration is missing from F1 variant, present on all other variants.

https://github.com/STMicroelectronics/STM32CubeF1/blob/5ddefcd6a64e793087436cc4ed42abd1886952c1/Drivers/STM32F1xx_HAL_Driver/Inc/stm32f1xx_hal.h#L59-L61

**To Reproduce** : By inspection

See as example:
https://github.com/STMicroelectronics/STM32CubeF4/blob/5d01400afd60410f6e049cbd19179a67d44d53fd/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal.h#L200-L207
and
https://github.com/STMicroelectronics/STM32CubeF7/blob/08376dce1b404687e4a86b73077f396bccfc9cb5/Drivers/STM32F7xx_HAL_Driver/Inc/stm32f7xx_hal.h#L195-L201

```
Searching 2003 files for "uwTick" (case sensitive, whole word)

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32F0xx_HAL_Driver\Inc\stm32f0xx_hal.h:
  531    * @{
  532    */
  533: extern __IO uint32_t uwTick;
  534  extern uint32_t uwTickPrio;
  535  extern HAL_TickFreqTypeDef uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32F2xx_HAL_Driver\Inc\stm32f2xx_hal.h:
  174    * @{
  175    */
  176: extern __IO uint32_t uwTick;
  177  extern uint32_t uwTickPrio;
  178  extern HAL_TickFreqTypeDef uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32F3xx_HAL_Driver\Inc\stm32f3xx_hal.h:
  889    * @{
  890    */
  891: extern __IO uint32_t uwTick;
  892  extern uint32_t uwTickPrio;
  893  extern HAL_TickFreqTypeDef uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32F4xx_HAL_Driver\Inc\stm32f4xx_hal.h:
  203    * @{
  204    */
  205: extern __IO uint32_t uwTick;
  206  extern uint32_t uwTickPrio;
  207  extern HAL_TickFreqTypeDef uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32F7xx_HAL_Driver\Inc\stm32f7xx_hal.h:
  197    * @{
  198    */
  199: extern __IO uint32_t uwTick;
  200  extern uint32_t uwTickPrio;
  201  extern HAL_TickFreqTypeDef uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32G0xx_HAL_Driver\Inc\stm32g0xx_hal.h:
  716    * @{
  717    */
  718: extern __IO uint32_t uwTick;
  719  extern uint32_t uwTickPrio;
  720  extern uint32_t uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32G4xx_HAL_Driver\Inc\stm32g4xx_hal.h:
  571    * @{
  572    */
  573: extern __IO uint32_t uwTick;
  574  extern uint32_t uwTickPrio;
  575  extern uint32_t uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32H7xx_HAL_Driver\Inc\stm32h7xx_hal.h:
  965    * @{
  966    */
  967: extern __IO uint32_t uwTick;
  968  extern uint32_t uwTickPrio;
  969  extern HAL_TickFreqTypeDef uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32L0xx_HAL_Driver\Inc\stm32l0xx_hal.h:
  355    * @{
  356    */
  357: extern __IO uint32_t uwTick;
  358  /**
  359    * @}

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32L1xx_HAL_Driver\Inc\stm32l1xx_hal.h:
  909    * @{
  910    */
  911: extern __IO uint32_t uwTick;
  912  extern uint32_t uwTickPrio;
  913  extern uint32_t uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32L4xx_HAL_Driver\Inc\stm32l4xx_hal.h:
  604    * @{
  605    */
  606: extern __IO uint32_t uwTick;
  607  extern uint32_t uwTickPrio;
  608  extern uint32_t uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32MP1xx_HAL_Driver\Inc\stm32mp1xx_hal.h:
  710    * @{
  711    */
  712: extern __IO uint32_t uwTick;
  713  extern uint32_t uwTickPrio;
  714  extern HAL_TickFreqTypeDef uwTickFreq;

C:\Users\geoma\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32WBxx_HAL_Driver\Inc\stm32wbxx_hal.h:
  629    * @{
  630    */
  631: extern __IO uint32_t uwTick;
  632  extern uint32_t uwTickPrio;
  633  extern uint32_t uwTickFreq;
```

ASELSTM, 2020-05-18:
Hi @geosmall,

The issue you pointed out has been confirmed, a fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2020-05-18:
ST Internal Reference: 86528

closed 2020-10-12 by ALABSTM