STMicroelectronics/STM32CubeF1 #14: stm32f1xx's HAL_DMA_PollForTransfer can't be used for polling

Tasssadar, 2020-06-14:
The `HAL_DMA_PollForTransfer` function of stm32f1xx misleadingly always finishes the current DMA transfer on timeout, whether it is complete or not. This means that you can't actually poll for whether the transfer is done, because the act of calling `HAL_DMA_PollForTransfer` will finish it.

Even if this is the intended behavior, please at the very least mention it in the documentation of the function.

**Expected behavior**
`HAL_DMA_PollForTransfer` does not set state to `HAL_DMA_STATE_READY` on timeout.

RKOUSTM, 2020-06-19:
Hi @Tasssadar,

Thank you for reporting this issue. You request will be forwarded to our development teams. I will let you updated once they provide me with their answer.

With regards,

RKOUSTM, 2020-07-01:
Hi @Tasssadar ,

Thank you for your contribution. Could you please confirm you are using the latest firmware version (STM32Cube FW F1 v1.8.0) and that the issue is systematically reproducible on your side?

With regards,

Tasssadar, 2020-07-01:
I'm using platform.io and not the official tools, but it is coded like that here in this repository in master: https://github.com/STMicroelectronics/STM32CubeF1/blob/master/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_dma.c#L502

RKOUSTM, 2020-07-06:
Hi @Tasssadar,

I have checked with our development teams. They mentioned that for stm32f1xx_hal_dma.c driver code, there is no problem. However, timeout is not a blocking DMA error, and it is normal to come back to Ready state.

In order to allow a better understanding of the problem, could you please send us a code snippet allowing our development teams to reproduce (if useful),  so that our development teams can help give.

Thank you for your contribution.

With regards,

Tasssadar, 2020-07-06:
Yes, of course it is not an error. That's why it makes no sense to cut the current DMA transaction short and mark the state as ready, instead of keeping it "busy".

Here's some pseudo code of what I was trying to do:

```cpp
// This method wants to start DMA transfer and wait for its completion
void dma() {
   HAL_DMA_Start(&dmaTxHandle,  ...);

    while(dmaTxHandle.State != HAL_DMA_STATE_READY) {
        HAL_DMA_PollForTransfer_Really(&dmaTxHandle, HAL_DMA_FULL_TRANSFER, 0);

    // DMA transfer is complete here! (except it's not)
}
```

RKOUSTM, 2020-07-08:
Hi @Tasssadar,

Thank you for all these details. According to our development teams, when timeout is reached, it is better to free the DMA by setting it to HAL_DMA_STATE_READY  in order to start another transfer (of higher priority) or repeat the failed one.  

Then, based on this point of view, there is no need to fix or update the code. Thank you for your comprehension.

Please allow me to close this issue now. Thank you again for your report.

With regards,

Tasssadar, 2020-07-08:
It is fine to have a function that does that of course, but the naming and documentation should clearly state that no matter what you do, `HAL_DMA_PollForTransfer` will always finish the DMA transfer instead of just checking the state (which is what polling means IMHO).

RKOUSTM, 2020-07-24:
Hi @Tasssadar,

First of all, I would like to thank you for your contribution.  However, according to our development teams, the ``Polling Transfer Mode`` should have a non-zero timeout duration. In your pseudo-code above, the function ``HAL_DMA_PollForTransfer_Really(&dmaTxHandle, HAL_DMA_FULL_TRANSFER, 0)`` must be updated with a Timeout duration different to zero to ensure complete transfer. Indeed, in case you need a fast timing for the DMA transfer, you can use a ``HAL_DMA_IRQHandler`` function for an interrupt request. So, thank you once more for your contribution. 

Please, allow me to close this issue. Thank you for your comprehension.

With regards,

closed 2020-07-24 by RKOUSTM