STMicroelectronics/STM32CubeF1 #17: gcc Startup files don't initialize stack pointer

ftapajos, 2020-06-18:
I'm not sure if that is an issue, because it seems too critical to be in the code for that long, but the stackpointer is not being set anywhere in the reset_Handler subrountine for any of the startup_stm32f1xxxx.s files in the gcc folder

ALABSTM, 2020-06-19:
Hi @ftapajos,

Thank you for your question. It seems you will find your answer here (#5). Do not hesitate in case you still face an issue.

With regards,

ALABSTM, 2020-07-22:
Hi @ftapajos,

Did you find the answer you are looking for here (#5)?

With regards,

ALABSTM, 2020-07-27:
Hi @ftapajos,

Please allow me to close this issue. You can reopen it or open a new one in case you face any problem related to this aspect.

With regards,

closed 2020-07-27 by ALABSTM