STMicroelectronics/STM32CubeF1 #18: Syntax failure in stm32f1xx_hal_tim.h

STMAndreas, 2020-06-19:
Using STM32F103C8T6 with STM32CubeIDE 1.3.0

During Migration from STM32Cube_FW_F1_V1.3.0 to STM32Cube_FW_F1_V1.8.0 the compiler found a failure in  stm32f1xx_hal_tim.h:
`../Application/User/IR_rec_drv.c: In function 'HAL_TIM_IC_CaptureCallback':
../Drivers/STM32F1xx_HAL_Driver/Inc/stm32f1xx_hal_tim.h:1745:106: error: expected ':' before ')' token
   (((__CHANNEL__) == TIM_CHANNEL_1) ? ((__HANDLE__)->Instance->CCER &= ~(TIM_CCER_CC1P | TIM_CCER_CC1NP))) :\
                                                                                                          ^
../Drivers/STM32F1xx_HAL_Driver/Inc/stm32f1xx_hal_tim.h:1472:5: note: in expansion of macro 'TIM_RESET_CAPTUREPOLARITY'
     TIM_RESET_CAPTUREPOLARITY((__HANDLE__), (__CHANNEL__));               \
     ^~~~~~~~~~~~~~~~~~~~~~~~~
../Application/User/IR_rec_drv.c:184:13: note: in expansion of macro '__HAL_TIM_SET_CAPTUREPOLARITY'
             __HAL_TIM_SET_CAPTUREPOLARITY(htim, TIM_CHANNEL_4,TIM_INPUTCHANNELPOLARITY_RISING);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~`

I could solve the problem by removing one bracket at the end of line 1745 
(macro TIM_RESET_CAPTUREPOLARITY):
original:
` (((__CHANNEL__) == TIM_CHANNEL_1) ? ((__HANDLE__)->Instance->CCER &= ~(TIM_CCER_CC1P | TIM_CCER_CC1NP))) :\
`
modified:
`  (((__CHANNEL__) == TIM_CHANNEL_1) ? ((__HANDLE__)->Instance->CCER &= ~(TIM_CCER_CC1P | TIM_CCER_CC1NP)) :\
`
That's definitely one bracket too much there.
I compared this macro now with earlier versions, first with STM32Cube_FW_F1_V1.3.0 but also with all other once between 1.3.0 and 1.8.0 and I found the failure happens in 1.8.0
Furthermore, maybe there is more damaged in this macro (TIM_RESET_CAPTUREPOLARITY).
There is also a change in line 1747 which isn't listed in any release note. The macro was never changed from V1.3.0 til V1.7.0 but in V1.8.0 
It looks like an accidental deletion of a part of the line. I cannot test the function of this macro and the change because my project doesn't use channel3 of this timer but it looks like a failure for me. Maybe there is anyone in the community which knows the function of this macro and can confirm or reject this.
Here is the comparison of the macro from V1.7.0 to V1.8.0:  
[image]

(In general there is a restructure of the whole file in V1.8.0 therefore the line numbers differs very much from earlier version to V1.8.0)

ASELSTM, 2020-06-22:
Hi @STMAndreas,

First of all, we would like to thank you for your contribution.

The `TIM_RESET_CAPTUREPOLARITY` macro is used to reset an already set input capture polarity on a given channel which is the active edge of the input signal. The active edge can then be rising, falling or both depending on the TIM instance and the channel. 

```
#define  TIM_INPUTCHANNELPOLARITY_RISING      0x00000000U                       /*!< Polarity for TIx source */
#define  TIM_INPUTCHANNELPOLARITY_FALLING     TIM_CCER_CC1P                     /*!< Polarity for TIx source */
#define  TIM_INPUTCHANNELPOLARITY_BOTHEDGE    (TIM_CCER_CC1P | TIM_CCER_CC1NP)  /*!< Polarity for TIx source */
```

Indeed, the  `TIM_RESET_CAPTUREPOLARITY` macro has been updated within release v1.8.0. In fact, input capture polarity on channel 3 can be whether rising or falling and never both edges. That's why the `TIM_CCER_CC3NP` was removed.

The compilation error, has been already spotted and fixed internally but not yet published. The fix will be available in the next release.

Since the issue is already fixed, please allow me to close it. Thank you for your comprehension.

With regards,

closed 2020-06-22 by ASELSTM

STMAndreas, 2020-06-22:
Hi @ASELSTM,
cool! Thanks for the answer and explanation. 
STM32F1 series is available since more than 15 years and there is still something to optimize 😀
Best Regards