STMicroelectronics/STM32CubeF1 #19: HAL_I2C_Master_Receive may unintentionally leave interrupts disabled on exit

kasjer, 2020-07-14:
Function **HAL_I2C_Master_Receive** looks like it can return with error while previously disabled interrupts are not enabled as they would be in no error case.

Line that blocks interrupts;
https://github.com/STMicroelectronics/STM32CubeF1/blob/5ddefcd6a64e793087436cc4ed42abd1886952c1/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_i2c.c#L1334

Line that returns without enabling interrupts:
https://github.com/STMicroelectronics/STM32CubeF1/blob/5ddefcd6a64e793087436cc4ed42abd1886952c1/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_i2c.c#L1349

ASELSTM, 2020-07-16:
Hi @kasjer,

First of all we would like to thank you for your contribution.

Actually, the point you raised has already been discussed and fixed internally since a while, before we receive your request. The fix will be made available in the frame of a future release.

We cannot share a date for the moment. Stay tuned and thank you once more for your contribution and for your patience.

With regards,

ASELSTM, 2020-07-16:
ST Internal Reference:  80693

closed 2020-10-12 by ALABSTM