STMicroelectronics/STM32CubeF1 #23: Race condition in USB PCD control endpoint receive ISR

daniel-starke, 2020-09-23:
**Describe the set-up**
Tested with STM32F103C8T6. Same code can be found for F0 series and others.

**Describe the bug**
Usually, the procedure to return control to the host after a handled SETUP packet in USB is done by receiving 0 byte on the OUT control endpoint. For PMA assisted chips this results in a) an interrupt which confirms the finished receive operation followed by b) an interrupt for the next SETUP packet. In case it takes quite long to handle the a) interrupt (due to low interrupt priority for example) the length information of the SETUP packet will be overwritten. This is the relevant part from the code:
https://github.com/STMicroelectronics/STM32CubeF1/blob/6ead386a08dea17f4dffbdbcd36c9979cb53588b/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_pcd.c#L2130-L2174
The situation was already improved in 1.8.0 by avoiding the call of the user callback function in case of such an a) interrupt (see line 2157). However, the root issue remains and can lead to a race condition were a SETUP packet is received just between line 2152 and 2172 and the received byte count being overwritten in line 2172. This behavior can be proven by checking `(PCD_GET_ENDPOINT(hpcd->Instance, PCD_ENDP0) & USB_EP_SETUP) != 0` at line 2171.  
I suggest to properly re-check the `USB_EP_SETUP` bit and avoid `PCD_SET_EP_RX_CNT()` and `PCD_SET_EP_RX_STATUS()` if set.

ALABSTM, 2020-11-17:
Hi @daniel-starke,

Thank you for this report. Both issue description and suggested solution are clear. I will forward it to our development teams and get back to you as soon as I get their feedback.

With regards,

ALABSTM, 2020-12-15:
Hi @daniel-starke,

Our development teams confirmed the issue and approved your proposal. A fix will be made available in a future release. The below code snippet shows how an extra check will be added just before line [2172](https://github.com/STMicroelectronics/STM32CubeF1/blob/6ead386a08dea17f4dffbdbcd36c9979cb53588b/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_pcd.c#L2172).

Thank you again for your contribution.

With regards,

```diff
+ if ((PCD_GET_ENDPOINT(hpcd->Instance, PCD_ENDP0) & USB_EP_SETUP) == 0U)
+ {
    PCD_SET_EP_RX_CNT(hpcd->Instance, PCD_ENDP0, ep->maxpacket);
    PCD_SET_EP_RX_STATUS(hpcd->Instance, PCD_ENDP0, USB_EP_RX_VALID);
+ }
```

ALABSTM, 2020-12-15:
ST Internal Reference: 98281

daniel-starke, 2020-12-15:
This only improves the behavior but does not provides a complete solution. That is why I suggested a properly re-check of the bit and not just a simple check. Higher priority interrupts may still break the intended behavior here.  
One could disable interrupts during that code block for example, if this implies only a little performance overhead.  
I haven't done any tests here so please re-evaluate the possible solutions.

ALABSTM, 2020-12-18:
Hi @daniel-starke,

Thank you for this clarification. I forwarded your concern and proposal to our development teams. I hope to get back to you soon enough.

With regards,

bouattay, 2020-12-21:
Hello Daniel-starke,
Let me explain the proposed solution, so assuming device get interrupted during setup out status phase and host is faster and send  the next setup packet. The fix here should ensure to do not update the Rx countx while USB_EP_SETUP bit is set. From device side the second setup packet is ACKed, host advances the state to either data phase or status phase with IN or OUT packet, the device which is still blocked by the higher IT treatment, should just NAK the IN/OUT packets 

if ((PCD_GET_ENDPOINT(hpcd->Instance, PCD_ENDP0) & USB_EP_SETUP) == 0U) is even aligned with your comment

>>I suggest to properly re-check the USB_EP_SETUP bit and avoid PCD_SET_EP_RX_CNT() and PCD_SET_EP_RX_STATUS() if set.

Regards,
Aymen

daniel-starke, 2020-12-21:
Hello Aymen,
Thank you for the re-evaluation. What I suspect is, that the initial issue will still arise if the device got stuck for a while in a higher priority interrupt which was just triggered before entering the line with `PCD_SET_EP_RX_CNT`, i.e. before the host sends the SETUP packet to the device. The device would fill the internal buffer, set the byte count, ACK the package and after that return from the higher priority interrupt to clear the byte count at the line in question. The setup message is lost and both sides are stuck in an inconsistent state.