STMicroelectronics/STM32CubeF1 #25: Conflicting types in USBD_LL_Transmit and USBD_LL_PrepareReceive

lead-free, 2020-11-22:
**Describe the set-up**
STM32F103C8T6 Microcontroller. Initialization code generated with STM32CubeMX. USB configured as Device CDC Class. When generating the code "add necessary files as reference" checked, that is cube does not copy the libraries into the project folder.

**Describe the bug**
In [usbd_conf_template.c](https://github.com/STMicroelectronics/STM32CubeF1/blob/master/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_conf_template.c) declaration of functions [USBD_LL_Transmit](https://github.com/STMicroelectronics/STM32CubeF1/blob/276c4ab9537e83b9897badb68e02d9c4c46c5c22/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_conf_template.c#L160) and [USBD_LL_PrepareReceive](https://github.com/STMicroelectronics/STM32CubeF1/blob/276c4ab9537e83b9897badb68e02d9c4c46c5c22/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_conf_template.c#L174) does not match with definitions of these functions in [stm32_mw_usb_device/Core/Inc/usbd_core.h](https://github.com/STMicroelectronics/stm32_mw_usb_device/blob/master/Core/Inc/usbd_core.h). Last argument has type `uint16_t size` in declaration and `uint32_t size` in definition. Causes compilation error.

**How To Reproduce**
1. Generate initialization code with STM32CubeMX enabling USB as Device CDC Class. Check "add necessary files as reference".
2. Clone necessary libraries into your project
```shell
git init
git submodule add https://github.com/ARM-software/CMSIS_5 libraries/CMSIS_5
git submodule add https://github.com/STMicroelectronics/cmsis_device_f1.git libraries/cmsis_device_f1
git submodule add https://github.com/STMicroelectronics/stm32f1xx_hal_driver.git libraries/stm32f1xx_hal_driver
git submodule add https://github.com/STMicroelectronics/stm32_mw_usb_device.git libraries/stm32_mw_usb_device
```
3. Compile project

**Additional context**
I tried to reproduce the issue on STM32L412 everything went fine `USBD_LL_Transmit` and `USBD_LL_PrepareReceive` matched.

ALABSTM, 2021-01-19:
Hi @stansotn,

Thank you for reporting. You are absolutely right about this point. There is a type mismatch. Actually, this is an issue found in version **v2.5.3** of the USB Device library and fixed in version **v2.6.0**.

Currently, the STM32CubeF1 firmware supports the **v2.5.3** of the USB Device library. Unfortunately there is no plan to upgrade to the **v2.6.0**. I will check further with our development teams and let you know should there be any upgrade planned.

Thank you for your comprehension and thank you again for having reported this point.

With regards,

ALABSTM, 2021-03-08:
Hi @stansotn,

I hope you are fine. Please find the latest USB Device library release within the [stm32_mw_usb_device](https://github.com/STMicroelectronics/stm32_mw_usb_device) repository. Version **v2.6.0** and even version **v2.7.1** are available and **fully compatible** with the v2.5.3.

The purpose from this repository is double:
1. to make updates and fixes available to users as soon as possible and get rid of delays introduced by any firmware package release.
2. to allow users to profit from updates and fixes in case there is no upgrade plan for firmware packages they are using (like it is the case of the STM32CubeF1).

All you have to do is to overwrite the content of your [STM32_USB_Device_Library](https://github.com/STMicroelectronics/STM32CubeF1/tree/master/Middlewares/ST/STM32_USB_Device_Library) folder with the one of the above-mentioned repository. I hope this addresses your need. Please let me know should you face any issue. Thank you.

With regards,

stansotn, 2021-03-09:
@ALABSTM,

Thank you for addressing the issue! I will update my projects' submodules to the latest release tags and let you know if there are any issues.

closed 2021-03-09 by stansotn