STMicroelectronics/STM32CubeF1 #26: error LL_GPIO_SetPinPull() function behavior in LL_GPIO_Init()

simaodangan, 2020-11-22:
hello!

I have used STM32CubeMX V6.1.0(STM32Cube FW_F1 V1.8.3) to generate STM32F103ZETx's gpio init code, and find that thegpio init function `LL_GPIO_Init()` will call `LL_GPIO_SetPinPull()`, even if when i set set pin as output mode.

That will cause a error set zero to GPIO->ODR, in such code:

```C
LL_APB2_GRP1_EnableClock(LL_APB2_GRP1_PERIPH_GPIOF);

LL_GPIO_SetOutputPin(GPIOF, LED0_Pin);

GPIO_InitStruct.Pin = LED0_Pin;
GPIO_InitStruct.Mode = LL_GPIO_MODE_OUTPUT;
GPIO_InitStruct.Speed = LL_GPIO_SPEED_FREQ_LOW;
GPIO_InitStruct.OutputType = LL_GPIO_OUTPUT_OPENDRAIN;
LL_GPIO_Init(GPIOF, &GPIO_InitStruct);
```

i think `LL_GPIO_SetPinPull()` shall be called only when the pin is setted to output mode.

ASELSTM, 2021-01-05:
Hi @simaodangan,

Referring to the reference manual  [RM0008](https://www.st.com/resource/en/reference_manual/cd00171190-stm32f101xx-stm32f102xx-stm32f103xx-stm32f105xx-and-stm32f107xx-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf) the **Table 20** shown below is describing the port bit configuration. Based on this table, in the **General purpose output mode**, configuring the ODR register to 0 or 1 is **mandatory** unlike the AF mode where it doesn’t matter to configure the ODR or not. Thus, the LL_GPIO_Init() should always call the LL_GPIO_SetPinPull() to configure the ODR.

[image]

Actually, STM32F1 GPIO is a little different from the others STM32 platform. There is no PUPDR register to configure the pull-up/down when INPUT mode is selected, it rather depends on the ODR register value as described in the Table 20.  Thus, to maintain the driver compatibility with the rest of STM32 platforms, the LL_GPIO_SetPinPull() is kept and used to set the PULL in the ODR register.

Actually, STM32F1 GPIO is a little different from the others STM32 platform. There is no PUPDR register dedicated to configure the pull-up/down when INPUT mode is selected, it rather depends on the ODR register value as described in the Table 20. Thus, to maintain the driver compatibility with the rest of STM32 platforms, the LL_GPIO_SetPinPull() is kept and used to set the PULL in the ODR register.

That said, to fix the issue, you should call LL_GPIO_SetOutputPin() after LL_GPIO_Init() and not before. The code will be then as follows: 

```diff
LL_APB2_GRP1_EnableClock(LL_APB2_GRP1_PERIPH_GPIOF);
-// init the pin as high level
-LL_GPIO_SetOutputPin(GPIOF, LED0_Pin);

//config the pin
GPIO_InitStruct.Pin = LED0_Pin;
GPIO_InitStruct.Mode = LL_GPIO_MODE_OUTPUT;
GPIO_InitStruct.Speed = LL_GPIO_SPEED_FREQ_LOW;
GPIO_InitStruct.OutputType = LL_GPIO_OUTPUT_OPENDRAIN;
LL_GPIO_Init(GPIOF, &GPIO_InitStruct);

+// init the pin as high level
+LL_GPIO_SetOutputPin(GPIOF, LED0_Pin);
```

Hope this would help you and thank you for your contribution.

With regards,

simaodangan, 2021-01-10:
Thank you for your explanation.

If I call `LL_GPIO_SetOutputPin()`  after `LL_GPIO_Init()`, the pin will generate a Low-Level pulse between Hi-Z and High-Level, which is a illegal status and may damage the hardware when a break between two function calls.

Actually, I use STM32Cube to generate the code,  if I select "GPIO mode" as "Output Open Drain", and "GPIO output level" as "High". It doesn't work, and the code generated by cube  is like:

```C
LL_APB2_GRP1_EnableClock(LL_APB2_GRP1_PERIPH_GPIOF);

//The config:  "GPIO output level" -> "High"
LL_GPIO_SetOutputPin(GPIOF, LED0_Pin);

GPIO_InitStruct.Pin = LED0_Pin;
GPIO_InitStruct.Mode = LL_GPIO_MODE_OUTPUT;
GPIO_InitStruct.Speed = LL_GPIO_SPEED_FREQ_LOW;
GPIO_InitStruct.OutputType = LL_GPIO_OUTPUT_OPENDRAIN;
LL_GPIO_Init(GPIOF, &GPIO_InitStruct);

//and here, the pin is set to LOW level by LL_GPIO_Init()
```

I solve this problem by selecting "GPIO Pull-ip/Pull-down" as "Pull-up". 
The new code is like:

```C
LL_APB2_GRP1_EnableClock(LL_APB2_GRP1_PERIPH_GPIOF);

//The config:  "GPIO output level" -> "High"
LL_GPIO_SetOutputPin(GPIOF, LED0_Pin);

//config the pin
GPIO_InitStruct.Pin = LED0_Pin;
GPIO_InitStruct.Mode = LL_GPIO_MODE_OUTPUT;
GPIO_InitStruct.Speed = LL_GPIO_SPEED_FREQ_LOW;
GPIO_InitStruct.OutputType = LL_GPIO_OUTPUT_OPENDRAIN;
//additional code
GPIO_InitStruct.Pull = LL_GPIO_PULL_UP;
LL_GPIO_Init(GPIOF, &GPIO_InitStruct);

//and here, the pin is set to HIGH level
```

I think the it is confused for the user of STM32Cube.

ASELSTM, 2021-02-08:
Hi @simaodangan,

Pleased that the problem has been solved. 

Actually, this is rather a Cube MX related issue and we don't treat aspects related to CubeMX tool at GitHub level, they are rather treated at ST Community level precisely within the [CubeMX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated section. You can create a new post to submit your issue. 

As this issue is not directly related to software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to ecosystem (CubeMX tool), please allow me to close it.

Thank you for your comprehension and thank you again for your contribution. We are looking forward to reading from you again.

With regards,

closed 2021-02-08 by ASELSTM