STMicroelectronics/STM32CubeF1 #28: BackupRegister macro at LL module

cnxiaochun, 2021-01-19:
stm32f1xx_ll_rtc.h

`
#define LL_RTC_BKP_DR11                    (0x0000000BU)
`

stm32f1xx_hal_rtc_ex.h

`
#define RTC_BKP_DR11                        0x00000010U
`

Two definitions are inconsistent

ALABSTM, 2021-02-09:
Hi @cnxiaochun,

Thank you for this report. You are right. There is clearly an issue with the macros of the **LL** driver. All macros listed below shall be aligned with their equivalents in the HAL driver as in the example hereafter.

```diff
-#define LL_RTC_BKP_DR11 (0x0000000BU)
+#define LL_RTC_BKP_DR11 (0x00000010U)
```

With regards,

https://github.com/STMicroelectronics/STM32CubeF1/blob/276c4ab9537e83b9897badb68e02d9c4c46c5c22/Drivers/STM32F1xx_HAL_Driver/Inc/stm32f1xx_ll_rtc.h#L142-L173

ALABSTM, 2021-02-09:
ST Internal Reference: 101139

RKOUSTM, 2021-06-08:
Hi @cnxiaochun, 

I hope you are fine.

The issue you reported has been fixed in the frame of version v1.8.4 of the STM32CubeF1 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-06-08 by RKOUSTM