STMicroelectronics/STM32CubeF1 #29: Funcrion LL_RTC_TIME_Init  confusion RTC_Format

AntonNeutron, 2021-02-05:
Hello!
I have a problem initializing RTC
For example, if you are using the BIN format.
function LL_RTC_TIME_Init, first of all the time is checked for correctness

 if (RTC_Format == LL_RTC_FORMAT_BIN) {
    assert_param (IS_LL_RTC_HOUR24 (RTC_TimeStruct-> Clock));
    assert_param (IS_LL_RTC_MINUTES (RTC_TimeStruct-> Minutes));
    assert_param (IS_LL_RTC_SECONDS (RTC_TimeStruct-> Seconds));
  }
Looks good

But then the inequality sign (! =) Is used
and the time is converted from BCD to BIN even if it is already in BIN format.

 if (RTC_Format            !=         LL_RTC_FORMAT_BIN) {
      counter_time = (uint32_t) (((uint32_t) RTC_TimeStruct-> Clock * 3600U) + \
                                ((uint32_t) RTC_TimeStruct-> Minutes * 60U) + \
                                ((uint32_t) RTC_TimeStruct-> Seconds));
      LL_RTC_TIME_Set (RTCx, counter_time);
    }
otherwise // (RTC_Format == LL_RTC_FORMAT_BIN)
{
      counter_time = (((uint32_t) (__ LL_RTC_CONVERT_BCD2BIN (RTC_TimeStruct-> Clock)) * 3600U) + \
                      ((uint32_t) (__ LL_RTC_CONVERT_BCD2BIN (RTC_TimeStruct-> Minutes)) * 60U) + \
                      ((uint32_t) (__ LL_RTC_CONVERT_BCD2BIN (RTC_TimeStruct-> Seconds))));
      LL_RTC_TIME_Set (RTCx, counter_time);
    }

 If you change the sign != to == then everything comes in the right place.

The same happens in LL_RTC_ALARM_Init

ALABSTM, 2021-02-10:
Hello @AntonNeutron,

Looks like we've missed that. Our apologies and thank you very much for having drawn out our attention upon. This bug will be logged into our internal database and a fix will be made available soon enough hopefully.

With regards,

ALABSTM, 2021-02-10:
ST Internal Reference: 101170

RKOUSTM, 2021-06-08:
Hi @AntonNeutron, 

I hope you are fine.

The issue you reported has been fixed in the frame of version v1.8.4 of the STM32CubeF1 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-06-08 by RKOUSTM