STMicroelectronics/STM32CubeF1 #30: Purpose a fix to STM32F105 startup codes for proper initialization

ttimasdf, 2021-02-25:
**Describe the set-up**
 * The board: A custom STM32F105RCT6 development board from Taobao. board description (website in Chinese): https://item.taobao.com/item.htm?id=556036697936 . I can upload the schematic here if necessary, but any board should be okay to reproduce.
 * IDE: VSCode + PlatformIO (Core 5.1.0)
   * Platform pack: ststm32 + my own support of STM32F105 series https://github.com/ttimasdf/platform-ststm32/tree/add-generic-stm32f105x
   * Arduino Core: stm32duino + my own support of STM32F105 series https://github.com/ttimasdf/Arduino_Core_STM32/tree/variant-generic-stm32f105x

**Describe the bug**
The `premain()` function provided by stm32duino is not properly called for HAL initialization. I've submitted a PR here stm32duino/Arduino_Core_STM32#1310 and been advised to also post the fix here. The only I can see is that it called something generated by the compiler. I refer to the startup code from lines below and submit the aforementioned PR. I'd be happy if anyone explain a little bit about why these lines are necessary for things to work 🙏 

https://github.com/STMicroelectronics/STM32CubeF1/blob/003dfc9e6c2e424c19a25df3934afaf7fce660d6/Drivers/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/startup_stm32f103xb.s#L92-L98

**How To Reproduce**

TODO: I'll add more detailed tomorrow 

**Additional context**

The patch is available at ttimasdf/Arduino_Core_STM32@acbd6cd7c6e71a26512fe684f1d01edaa34deb70

ttimasdf, 2021-02-25:
Should I also submit an issue or PR here? https://github.com/STMicroelectronics/cmsis_device_f1

ALABSTM, 2021-02-26:
Hi @ttimasdf,

Thank you for this contribution. It will be logged into our internal database and a fix will be made available in the frame of a future release.

In case you would like to create a p-r, it would be preferable that you push it to this repository to allow linking it to this issue. If you prefer to push it to the [cmsis_device_f1](https://github.com/STMicroelectronics/cmsis_device_f1) though, you can. I will transfer this issue there and link your p-r to it.

Regarding your question, I will try to provide you with an answer as soon as possible.

With regards,

ALABSTM, 2021-02-26:
ST Internal Reference: 102149

RKOUSTM, 2021-06-08:
Hi @ttimasdf,

I hope you are fine.

The issue you reported has been fixed in the frame of version v1.8.4 of the STM32CubeF1 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-06-08 by RKOUSTM