STMicroelectronics/STM32CubeF1 #32: stm32f1xx_hal_rtc.c -- inefficient code in HAL_RTC_Init() function

a5021, 2021-06-18:
The function HAL_RTC_Init() implemented in stm32f1xx_hal_rtc.c source file (v.1.8.0) has minor issue with using macro MODIFY_REG() against WRITE-ONLY registers RTC->PRLH and RTC->PRLL (see lines 381 & 382). It looks a bit odd and using WRITE_REG() macro instead of MODIFY_REG() could be a fix.

ALABSTM, 2021-08-02:
Hi @a5021,

Thank you for this report. Your point is correct as the concerned registers are write-only as you mentioned. The lines eventually to be updated are:

https://github.com/STMicroelectronics/STM32CubeF1/blob/c750eab6990cac35ab05020793b0221ecc1a8ce5/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_rtc.c#L380-L382

And the update could be:

```diff
-    MODIFY_REG(hrtc->Instance->PRLH, RTC_PRLH_PRL, (prescaler >> 16U));
-    MODIFY_REG(hrtc->Instance->PRLL, RTC_PRLL_PRL, (prescaler & RTC_PRLL_PRL));
+    WRITE_REG(hrtc->Instance->PRLH, ((prescaler >> 16U) & RTC_PRLH_PRL));
+    WRITE_REG(hrtc->Instance->PRLL, (prescaler & RTC_PRLL_PRL));
```

Your report will be forwarded to our development teams. Thank you again.

With regards,

ALABSTM, 2021-08-02:
ST Internal Reference: 111431

RJMSTM, 2023-04-14:
Hello @a5021 

Thank you again for having reported this point. It has been fixed in the frame of version 1.8.5 of this firmware.

With regards,

closed 2023-04-27 by RJMSTM