STMicroelectronics/STM32CubeF1 #33: Bad modification in stm32f1xx_hal_adc_ex.c

CanastraRF, 2021-08-01:
In the newest version of stm32f1xx_hal_adc_ex.c in function

HAL_ADCEx_MultiModeStart_DMA()
HAL_ADCEx_MultiModeStop_DMA()
HAL_ADCEx_MultiModeConfigChannel()

ADC_HandleTypeDef tmphadcSlave={0};
is added.
This create a const in flash of ADC_HandleTypeDef and copy this to stack.
A better solution is somthing like
memset(&tmphadcSlave, 0, sizeof(tmphadcSlave))

Improvment:
No const in flash,

ASELSTM, 2021-08-20:
ST Internal Reference: 112128

ASELSTM, 2023-04-07:
Hi @CanastraRF,

Our development team has unfortunately declined your proposal. According to them, the use of `memset()` would imply the use of C library within the HAL/LL drivers, which is not thread safe.

Please allow me thus to close this thread. Thank you for your comprehension.

With regards,

closed 2023-04-07 by ASELSTM