STMicroelectronics/STM32CubeF1 #37: MMC HAL driver fails to init

jhansen3141, 2021-10-07:
The current MMC HAL driver fails to init due to not selecting the card before attempting to read the extended CSD register. 

In `MMC_InitCard `in stm32f1xx_hal_mmc.c, the `HAL_MMC_GetCardCSD `function is called which in turn calls `MMC_ReadExtCSD `attempting to read the extended CSD register. However, this is done before the call to select the card which causes the `MMC_ReadExtCSD `function/command to fail.

**Current code in `MMC_InitCard`:**
```
  /* Get CSD parameters */
  if (HAL_MMC_GetCardCSD(hmmc, &CSD) != HAL_OK)
  {
    return hmmc->ErrorCode;
  }

  /* Select the Card */
  errorstate = SDMMC_CmdSelDesel(hmmc->Instance, (uint32_t)(((uint32_t)hmmc->MmcCard.RelCardAdd) << 16U));
  if(errorstate != HAL_MMC_ERROR_NONE)
  {
    return errorstate;
  }
```

**Change needed for fix in `MMC_InitCard`:**
```
  /* Select the Card */
  errorstate = SDMMC_CmdSelDesel(hmmc->Instance, (uint32_t)(((uint32_t)hmmc->MmcCard.RelCardAdd) << 16U));
  if(errorstate != HAL_MMC_ERROR_NONE)
  {
    return errorstate;
  }

  /* Get CSD parameters */
  if (HAL_MMC_GetCardCSD(hmmc, &CSD) != HAL_OK)
  {
    return hmmc->ErrorCode;
  }

```

RKOUSTM, 2021-10-19:
ST Internal Reference: 110782

RKOUSTM, 2021-10-19:
Duplicate of [STM32CubeF4#73](https://github.com/STMicroelectronics/STM32CubeF4/issues/73).

RJMSTM, 2023-04-27:
Hello @jhansen3141 

Thank you again for having reported this point. It has been fixed in the frame of version 1.8.5 of this firmware.

With regards,

closed 2023-04-27 by RJMSTM