STMicroelectronics/STM32CubeF1 #38: MMC DMA Reads/Writes Fail to Update DMA Direction

jhansen3141, 2021-10-13:
In stm32f1xx_hal_mmc.c, `HAL_MMC_ReadBlocks_DMA `and `HAL_MMC_WriteBlocks_DMA`, the DMA direction is set in the DMA CCR register before starting the DMA transfer. However, this register cannot be updated without first disabling DMA (clearing the EN bit in the CCR register). This is not currently done in the driver and therefore the DMA direction is never updated between MMC reads and writes.
```
hmmc->hdmatx->Init.Direction = DMA_MEMORY_TO_PERIPH;
MODIFY_REG(hmmc->hdmatx->Instance->CCR, DMA_CCR_DIR, hmmc->hdmatx->Init.Direction); // Fails to set because EN bit is set
```

```
__HAL_DMA_DISABLE(hmmc->hdmatx);
hmmc->hdmatx->Init.Direction = DMA_MEMORY_TO_PERIPH;
MODIFY_REG(hmmc->hdmatx->Instance->CCR, DMA_CCR_DIR, hmmc->hdmatx->Init.Direction); // Works because EN bit was cleared above
```

This issue can be overcome by having the application code disable DMA (call `__HAL_DMA_DISABLE(hdma)` before calling either `HAL_MMC_WriteBlocks_DMA `or `HAL_MMC_ReadBlocks_DMA`. However, the driver should be fixed so this is not required.

RKOUSTM, 2021-12-20:
Hi @jhansen3141,

Thank you for your contribution. The reported point will be forwarded to our development teams. I will get to you as soon as they provide their feedback.

Thank you again for your contribution and for your patience.

With regards,

ALABSTM, 2022-01-30:
Hi @jhansen3141,

Thank you for this report. We will get back to you with a feedback as soon as possible. Please excuse the delay it may take us sometimes to reply. Thank you for your comprehension.

With regards,

RKOUSTM, 2022-02-28:
Hi @jhansen3141,

Thank you for your contribution. An update of the MMC HAL driver will take this problem into into account and it will be made available in the frame of a future release v1.8.5.

Thank you again for this contribution and thank you in advance for your patience.

With regards,

RKOUSTM, 2022-04-15:
ST Internal Reference: 110782

RJMSTM, 2023-04-14:
Hello @jhansen3141 

Thank you again for having reported this point. It has been fixed in the frame of version 1.8.5 of this firmware.

With regards,

closed 2023-04-27 by RJMSTM