STMicroelectronics/STM32CubeF1 #39: missing static in HAL_RCC_GetSysClockFreq()

CanastraRF, 2022-01-03:
Hi
In the function

`uint32_t HAL_RCC_GetSysClockFreq(void)
`
are five const array

```
#if defined(RCC_CFGR2_PREDIV1SRC)
  uint8_t const aPLLMULFactorTable[14U] = {0U, 0U, 4U, 5U, 6U, 7U, 8U, 9U, 0U, 0U, 0U, 0U, 0U, 13U};
  uint8_t const aPredivFactorTable[16U] = {1U, 2U, 3U, 4U, 5U, 6U, 7U, 8U, 9U, 10U, 11U, 12U, 13U, 14U, 15U, 16U};
#else
  uint8_t const aPLLMULFactorTable[16U] = {2U, 3U, 4U, 5U, 6U, 7U, 8U, 9U, 10U, 11U, 12U, 13U, 14U, 15U, 16U, 16U};
#if defined(RCC_CFGR2_PREDIV1)
  uint8_t const aPredivFactorTable[16U] = {1U, 2U, 3U, 4U, 5U, 6U, 7U, 8U, 9U, 10U, 11U, 12U, 13U, 14U, 15U, 16U};
#else
  uint8_t const aPredivFactorTable[2U] = {1U, 2U};
#endif /*RCC_CFGR2_PREDIV1*/

```
they should bee static

Please fix this

RKOUSTM, 2022-01-07:
Hi @CanastraRF,

Thank you for your report. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you again for you report.

With regards,

RKOUSTM, 2022-01-07:
ST Internal Reference: 120495

RJMSTM, 2023-04-14:
Hello @CanastraRF 

Thank you again for having reported this point. It has been fixed in the frame of version 1.8.5 of this firmware.

With regards,

closed 2023-04-27 by RJMSTM