STMicroelectronics/STM32CubeF1 #41: HID code is not standards compliant

jmgurney, 2022-02-18:
**Describe the bug**

If an implementation using either the custom hid device class or the hid device class, and issues a GET_REPORT request on the control endpoint, an error will be returned instead of the required response.

The code here:
https://github.com/STMicroelectronics/STM32CubeF1/blob/master/Middlewares/ST/STM32_USB_Device_Library/Class/CustomHID/Src/usbd_customhid.c#L445

Is clearly missing a case for CUSTOM_HID_REQ_GET_REPORT.

Page 51 of the USB HID v1.11 standard has a note that this is mandatory and must be supported by all devices.

Page 25 of UM1734 (§ 5.2 claims that the HID class implementation in standards compliant when it is clearly not.

ALABSTM, 2022-04-12:
Hi @jmgurney,

Thank you for having reported this point. Actually, version **2.11.0** of the USB Device library has just been released, which addresses your point, as you can see [here](https://github.com/STMicroelectronics/stm32_mw_usb_device/blob/master/Class/CustomHID/Src/usbd_customhid.c#L406).

You can get it from the [stm32_mw_usb_device](https://github.com/STMicroelectronics/stm32_mw_usb_device) repo and use its content to overwrite the same files in your STM32CubeF1 firmware package, as this version is **fully compatible** with older ones.

Unfortunately, there is no plan in the near future to integrate this version into the STM32CubeF1 firmware. We will notify you, should there be any update. Please allow me to close this issue. Thank you for your comprehension.

With regards,

jmgurney, 2022-04-25:
Please keep this issue open until the firmware package has been updated. The reason I want this to remain open is so that other people who encounter this issue do not have to spend as long trying to figure out what is wrong with their devices.

I do understand that you likely want to keep the number of issues that are open low on your package, but IMO, that should just encourage your management to update and properly maintain the firmware for the micro controllers that you sell, and claim are standards compliant.