STMicroelectronics/STM32CubeF1 #46: CMSIS-Pack for STM32F10x series?

jankap, 2022-05-05:
Hi all,

while there are CMSIS-Packs (see https://www.keil.com/dd2/pack/) for almost all ST series, there is none for STM32F10x. Same situation when creating a project in STM32CubeMX; the sources are manually added to MDK instead of using any CMSIS-Pack.

Is there a specific reason behind it?

Thanks :)

jankap, 2022-06-14:
Dear @ALABSTM, does the `To do` assignment mean, that you or your team will work on a pack? I was going to create it on my own but of course I'll wait if there are plans like that.

Thanks!

ALABSTM, 2022-08-31:
Dear @jankap,

First of all, please excuse this delayed reply. Sometimes, it takes some time to answer.

I checked with our development teams and they said this issue is out of our scope. Indeed, the CMSIS packs available on Keil's [webpage](https://www.keil.com/dd2/pack/) are managed by their teams. You can reach out to their support [here](https://www.keil.com/support/).

As this topic is out of our scope, please allow me to close this thread. Thank you for your comprehension.

With regards,

closed 2022-08-31 by ALABSTM