STMicroelectronics/STM32CubeF1 #49: no memory pool implementation in cmisi_os2.c

xhlinux, 2022-06-10:
such as：osMemoryPoolNew func is not implement, but defined in cmsis_os2.h

```
//  ==== Memory Pool Management Functions ====

/// Create and Initialize a Memory Pool object.
/// \param[in]     block_count   maximum number of memory blocks in memory pool.
/// \param[in]     block_size    memory block size in bytes.
/// \param[in]     attr          memory pool attributes; NULL: default values.
/// \return memory pool ID for reference by other functions or NULL in case of error.
osMemoryPoolId_t osMemoryPoolNew (uint32_t block_count, uint32_t block_size, const osMemoryPoolAttr_t *attr);
```

and in STM32Cube_FW_F1_V1.8.4\Middlewares\Third_Party\FreeRTOS\Source\CMSIS_RTOS_V2\cmsis_os1.c
```
#if (defined(osFeature_Pool) && (osFeature_Pool != 0))&& !defined(os1_Disable_Pool)

osPoolId osPoolCreate (const osPoolDef_t *pool_def) {

  if (pool_def == NULL) {
    return NULL;
  }
  return osMemoryPoolNew(pool_def->pool_sz, pool_def->item_sz, &pool_def->attr);
}
```
but  osMemoryPoolNew can't link，

so where is osMemoryPoolNew defined？

HBOSTM, 2022-07-15:
ST Internal Reference: 131766

HBOSTM, 2023-04-04:
Hi @xhlinux ,

The osMemoryPoolNew() API shall be defined by user application as requested in the cmsis_os2.h 
It is a function wrapper and up to user to define the memory used in his application.
Otherwise we aren't owner on FreeRTOS MW stack, So please allow me to close this issue and thank you for your contribution.

With Regards,

closed 2023-04-04 by HBOSTM

ALABSTM, 2025-04-25:
See also [STM32CubeF4#170](https://github.com/STMicroelectronics/STM32CubeF4/issues/170)