STMicroelectronics/STM32CubeF1 #50: Error template with FatFs syscall.c

Coloryr, 2022-06-14:
Pack Version: 1.8.4

The generated template file cannot be used with FreeRTOS.
This problem occurs in the F1 chip. The H7 chip is normal.
F1:
[image]

H7:
[image]

HBOSTM, 2023-02-10:
Hello @Coloryr ,

This point has been reported to our development teams for deeper analysis. We will get back to you as soon as we have updates.

With Regards,

HBOSTM, 2023-02-10:
ST Internal Reference: 145227

RJMSTM, 2023-04-14:
Hello @Coloryr 

There is no plan to upgrade MW for legacy STM32 series.

With Regards,
Rania

closed 2023-04-14 by RJMSTM