STMicroelectronics/STM32CubeF1 #58: HAL_RTC_Init() looses time

net147, 2023-04-19:
Same issue as https://github.com/STMicroelectronics/STM32CubeH7/issues/24. HAL_RTC_Init() is called on each reset which loses up to a second of time.

RJMSTM, 2023-04-20:
Hello @net147,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension. 

With regards,

ALABSTM, 2023-05-26:
ST Internal Reference: 126464

ALABSTM, 2023-05-26:
Hi @net147,

Thank you for this report. We are already studying a solution to this issue as STM32F1xx microcontrollers support a different version of RTC than supported by other STM32 microcontrollers.

We will keep you updated.

With regards,

ALABSTM, 2023-08-01:
Hi @net147,

I hope you are fine. Back to you about this point you reported. Our development teams suggested, in the particular case of the STM32F1 series, the time loss shall be handled **at user application's level** rather than at HAL RTC driver's.

As explained in other posts, **during the RTC initialization**, the counter is **frozen**. Thus, the observed time loss. RTC initialization happens when executing  the `HAL_RTC_Init()` function. This function shall be executed **only once**, after a power reset for instance, as the RTC is not yet initialized or needs to be reinitialized.

You can refer to [this](https://github.com/STMicroelectronics/STM32CubeF1/tree/master/Projects/STM32VL-Discovery/Examples/RTC/RTC_LowPower_STANDBY) example showing **how to use the backup registers** to avoid executing  the `HAL_RTC_Init()` function in case the RTC peripheral has been already initialized.

I hope this helps. Please allow me to close this issue. Thank you for your comprehension and thank you for your report again.

With regards,

closed 2023-08-01 by ALABSTM