STMicroelectronics/STM32CubeF1 #59: USB keyboard. LED control

alekseibutiaev, 2023-05-20:
Custom board and with MCU STM32F10[5-7]
the function USBH_HID_SetReport does not works correctly.

**How To Reproduce**
1. Indicate the global behavior of your application project.
 works with USB keyboard
  write led status but does not changed.. but changed after press some keys

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
  module MW

3. The use case that generates the problem.

4. How we can reproduce the problem.
 write status of led by function USBH_HID_SetReport

**Additional context**
i have solution 
```

TOUNSTM, 2023-05-24:
Hello @alekseibutiaev,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,

alekseibutiaev, 2023-07-13:
i found it in this file

ALABSTM, 2023-08-02:
ST Internal Reference: 158663

closed 2025-11-13 by ALABSTM