STMicroelectronics/STM32CubeF1 #64: Compiler warning for unused parameters in stm32f1xx_ll_adc.h

IvanSojic, 2023-08-11:
Prevent unused argument compilation warnings (with some compilers) when using stm32f1xx_ll_adc.h

**Additional context**
* A solution would be to use UNUSED macro
[image]

RJMSTM, 2023-08-15:
Hello @IvanSojic,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With Regards,

RJMSTM, 2023-10-04:
Hello @Ivan,

Please can you precise the flags used in your example for better analysis? 

Thanks,
Rania

IvanSojic, 2023-10-28:
Hi @RJMSTM 
Currently we have enabled "-Wall -Wextra" for gcc compiler.
Best regards, Ivan

RJMSTM, 2023-11-28:
Hello,
for LL drivers you can use in stm32f1xx_ll_adc.h 
 
` (void)(ADCxy_COMMON);`

Regards,
Rania

RJMSTM, 2024-06-25:
ST Internal Reference: 184950

TOUNSTM, 2024-08-02:
Fixed in 42472d601caa457b02d849c7de5f9792629ee324

closed 2024-08-02 by TOUNSTM