STMicroelectronics/STM32CubeF1 #68: F1 series chips lack freertos_mpool.h and freertos_os2.h files

keysking, 2024-08-17:
After upgrading the F1 series package of STM32CubeIDE to version 1.8.6, I noticed that in newly created projects, the cmsis_os2.c file now includes references to freertos_mpool.h and freertos_os2.h. However, these two files are missing from the project.
[image]

I managed to locate these files in the 1.8.6 package, but even after placing them in the project's CMSIS_RTOS_V2 directory, I encountered more errors.
[image]

TOUNSTM, 2024-08-19:
ST Internal Reference: 188615

TOUNSTM, 2024-08-19:
Hello @keysking,

We confirm the abnormal behavior you reported. It has been reported to our development teams. A fix will be available in the frame of STM32CubeMX release 6.13.0. As there is no impact on the software provided within this repository, please allow me to close this issue.

Thank you for your comprehension.
With regards,

closed 2024-08-19 by TOUNSTM