STMicroelectronics/STM32CubeF1 #69: 1.8.6 has a bug

18268159377, 2024-09-05:
FreeRTOS\Source\CMSIS_RTOS_V2\cmsis_os2.c:33:10: fatal error: freertos_mpool.h: No such file or directory

TOUNSTM, 2024-09-06:
See Also https://github.com/STMicroelectronics/STM32CubeF1/issues/68

TOUNSTM, 2024-09-06:
See Also https://github.com/STMicroelectronics/stm32_mw_freertos/issues/5

TOUNSTM, 2024-09-06:
Hello @18268159377,

Thank you for your report. This point is related to the CubeMX generation problem and not directly related to the firmware published in this repository. A fix will be available in the frame of STM32CubeMX release 6.13.0. As there is no impact on the software provided within this repository, please allow me to close this issue.

Thank you for your comprehension.
With regards,

closed 2024-09-06 by TOUNSTM

Coloryr, 2024-09-18:
How to get the `STM32CubeMX release 6.13.0`?

konrin, 2024-11-08:
@TOUNSTM Hi! How to get around this problem without waiting for the IDE release? I already need to work with cmsis v2.
Could you please provide steps to solve this problem?