STMicroelectronics/STM32CubeF1 #72: 为什么驱动文件里面的内容是空的？？？

qq3521810385, 2025-01-17:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
 * IDE or at least the compiler and its version.

**Additional context**
If you have a first analysis or a patch proposal, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

ALABSTM, 2025-01-20:
Hi @qq3521810385,

Would you please rephrase your question using English language?

With regards,

ALABSTM, 2025-01-22:
Hi @qq3521810385,

Please allow me to close this issue. You can log another one if you want, in **English** please.

With regards,

closed 2025-01-22 by ALABSTM